ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #1: Dataset for testing

USTfgaoaa, 2020-05-09:
Thanks for sharing the code. 

I'm interested in trying your method, could you provide a dataset for testing? I see now the link to the dataset is empty.

Regards.

ChaoqinRobotics, 2020-05-09:
Thanks for your interest.

You can try the dataset I attach in this email, and run it by "rosbag play
lidar_imu_dataset --clock". If you want to run your own dataset, please
make sure to set the extrinsic transform between lidar and IMU by hand.

If you have any questions, please feel free to contact me.

Kind regards,
 lidar_imu_dataset.zip
<https://drive.google.com/file/d/19UUcO77L-g-RsZd_SLyr6O39S2JXJtiK/view?usp=drive_web>

closed 2020-05-10 by USTfgaoaa

G3tupup, 2020-05-12:
> Thanks for your interest. You can try the dataset I attach in this email, and run it by "rosbag play lidar_imu_dataset --clock". If you want to run your own dataset, please make sure to set the [...]

It seems that I don't have the authority to get the dataset. What's the reason