ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #3: State update

G3tupup, 2020-05-12:
Hi, I'm interested in your robocentric way to realize the state update. According to your paper and code, v_new = r * v_old, g_new = r * g_old. In the meanwhile, the covariance of v and g do not change. Is it a better way if I update the covariance by : cov_v_new = r * cov_v_old * rt, cov_g_new =  r * cov_g_old * rt, which seems to be more appropriate? Thank you!

ChaoqinRobotics, 2020-05-13:
Thank you for pointing this out! You are right. Given the covariance of x, cov, the covariance of z = Ax should be A*cov*At. I will update it in the next version. Thank you very much! : )

G3tupup, 2020-05-13:
> Thank you for pointing this out! You are right. Given the covariance of x, cov, the covariance of z = Ax should be A_cov_At. I will update it in the next version. Thank you very much! : )

You are welcome~

closed 2020-05-13 by G3tupup