ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #4: Ft matrix

G3tupup, 2020-05-13:
In your code in KalmanFilter.hpp line 159: 
Ft.block<3, 3>(GlobalState::att_, GlobalState::att_) =
          skew(gyr - state_tmp.bw_);
But according to your paper and code in LIO-Mapping, this block is  "-skew(gyr - state_tmp.bw_)", Is this a bug?

ChaoqinRobotics, 2020-05-13:
Thanks you for pointing it out! It's a typo in the paper. Please follow the code. If you are interested in the derivation, you can refer to "Quaternion kinematics for the error-state Kalman lter", where you can find similar formulas (but the definition of gravity is a littble bit different, which leads to different values in the gravity block in Ft). : )

ChaoqinRobotics, 2020-05-13:
It is a bug. I have revised the code and test it in all datasets. It seems like it does not has any influences to the result. I think it is because this term is so small compared to other blocks that its effect is not obvious. Anyways, thanks for point it out : )

Also, please notice that there is a new typo in my paper abuot Ft.  The derivative of the velocity with respect to the gravity should be a 3x3 identity matrix, while the derivative of the attitude with respect to the gravity should be leaved blank, just like what the code does--Ft.block<3, 3>(GlobalState::vel_, GlobalState::gra_) = M3D::Identity();

G3tupup, 2020-05-14:
> It is a bug. I have revised the code and test it in all datasets. It seems like it does not has any influences to the result. I think it is because this term is so small compared to other blocks [...]

Thank you! It seems to be the same as typos in Joan Sola's paper.

closed 2020-05-14 by G3tupup

narutojxl, 2020-05-26:
It seems that coefficient **A** of equ 375 in joan sola is right.
[image]
BTW, in lins paper equ 10, 11, the rotation matrix's superscript should be none or $b_k$, is it a typo?
[image]