ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #6: jacobian meaning in integrationBase.h

narutojxl, 2020-05-22:
Hi doctor qin, thanks for sharing the code! 
when preintegrating imu measurements, What's the [jacobian](https://github.com/ChaoqinRobotics/LINS---LiDAR-inertial-SLAM/blob/master/lins/include/integrationBase.h#L83) meaning, is there some reference paper to read first to understand, thanks very much!
BTW, i see the functions like estimateInitialState1(), estimateInitialState2(), estimateInitialState3() and solveGyroscopeBias() in [StateEstimator.hpp](https://github.com/ChaoqinRobotics/LINS---LiDAR-inertial-SLAM/blob/master/lins/include/StateEstimator.hpp) are not called by someone.

closed 2020-05-26 by narutojxl

ChaoqinRobotics, 2020-05-26:
Sorry for replying late. You can find the definition of this jacobina from VINS:Mono by Dr. Tong Qin. These functions are  for ba, bg estimation in the initialization step. But their performances are not so satisfying. I would update them in the future.

narutojxl, 2020-05-26:
Thanks a lot :) :)