ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #7: distance of point to line wrt point jacobian

narutojxl, 2020-05-26:
Hi doctor qin, when we calculating the distance of point P0 to line(P1 - P2) wrt P0 jacobian, the jacobian why is [this form](https://github.com/ChaoqinRobotics/LINS---LiDAR-inertial-SLAM/blob/master/lins/include/StateEstimator.hpp#L1044)?  
Let we 
[image]
How you get the formal like above,thanks very very much !

ChaoqinRobotics, 2020-05-28:
Thanks for asking. I would provide a detailed derivation in the future. A littble bit busy in recent month : )

narutojxl, 2020-05-28:
Thanks very much, looking forward!

closed 2020-05-28 by narutojxl

narutojxl, 2020-06-10:
Sorry, my mistake for writing the error formua of  d, it shoule be like this:
[image]
The jacobian in code is correct, thanks @qin :)