ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #8: residual wrt translation jacobian

narutojxl, 2020-05-27:
Hi doctor qin, when we calculating residual(point to line and point to plane) wrt translation jacobian in 
[translation jacobian in calculateTransformation()](https://github.com/ChaoqinRobotics/LINS---LiDAR-inertial-SLAM/blob/master/lins/include/StateEstimator.hpp#L1250) and [translation jacobian in performIESKF()](https://github.com/ChaoqinRobotics/LINS---LiDAR-inertial-SLAM/blob/master/lins/include/StateEstimator.hpp#L530), it seems that there miss a  variable `s`?
>  double s =  (1.f / SCAN_PERIOD) * (keypoint.intensity - int(keypoint.intensity));
Is it right? Thanks very much !

ChaoqinRobotics, 2020-05-28:
s is for undistorting motion distortion. After doing it, we assume the all points are collected in bk/bk+1 frame. As a consequence, the optimization step do not need to worry abuot the motion distortion, and we do not need to care for s in jacobian calculation

narutojxl, 2020-05-28:
But the weight of every undistorted point contributing to residual jacobian is different. I think as following: 
[image]
Knowing you are very busy recently, please have a look when you are free. 
Best regards!

closed 2020-05-28 by narutojxl