ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #9: filter's state

narutojxl, 2020-05-27:
Hi doctor qin,  filter's state in code [KalmanFilter.hpp](https://github.com/ChaoqinRobotics/LINS---LiDAR-inertial-SLAM/blob/master/lins/include/KalmanFilter.hpp#L118) seems is $X^{bk}_{b_{k+1}}$. But in the lins paper, the filter's state is $\delta X$, which is the error state vector of  $X^{b_k}_{b_{k+1}}$, why?
In joan sola's paper, the filter's state is the error state and the R related matrix of equ 237 is $R_t$, which is the rotation in G frame, does not agree with R related matrix of equ 9-11 in lins paper, which is  $R^{b_k}_t$ rotation from time *t* to  $b_{k}$, why?
Thanks very much!

ChaoqinRobotics, 2020-05-28:
Thanks for asking. Indeed, $X^{bk}{b{k+1}}$ denotes the error state mentioned in our paper. It is just because we have not change the notation in the code.

narutojxl, 2020-05-28:
Thank very much! :)

closed 2020-05-28 by narutojxl