ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #10: imu extrinsics are not used

shishiteng, 2020-05-28:
I searched for two variables `init_rbl` and `init_tbl` in the code but found that they were not used actually. Is this a bug?

narutojxl, 2020-05-28:
In  [imu callback() ](https://github.com/ChaoqinRobotics/LINS---LiDAR-inertial-SLAM/blob/master/lins/src/lib/Estimator.cpp#L132), author thinks the roll and pitch degrees between imu and laser are zero, the yaw from laser to imu  is  **imu_misalign_angle** in config file.

ChaoqinRobotics, 2020-05-28:
Thanks for your question. LINS program here is just a simplified version. These two values are used in the full program. We will use init_rbl and init_tbl  in the following update.

shishiteng, 2020-05-28:
> Thanks for your question. LINS program here is just a simplified version. These two values are used in the full program. We will use init_rbl and init_tbl in the following update.

Got it! Looking foward the new update!

closed 2020-05-28 by shishiteng

bryantaoli, 2021-06-27:
> Thanks for your question. LINS program here is just a simplified version. These two values are used in the full program. We will use init_rbl and init_tbl in the following update.

When will there be an update?