ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #11: imu bias noise unit

narutojxl, 2020-06-04:
Hi doctor qin,
In lins paper, imu bias's derivates **n_ba,  n_bg** unit  are **m/(s^2)/s** and **rad/s/s** repectively? 
In cofig file exp_port.yaml:
* **acc_w** is **n_ba**.   **acc_w**'s unit is  **ug/s = 1e-6 * 9.81 m/(s^2)/s**. 
*  **gyr_w** is **n_bg**.  **gyr_w**'s unit is **degree/sqrt(h) = pi/180.0/sqrt(3600) rad/sqrt(s)**, which is not consistent with the paper? 

I'm confused with the imu bias noise unit,  please have a look when you are free, thanks a lot !

narutojxl, 2020-06-15:
Hi @qin, Could you give a description in README about the default units of  imu noise used in code ?   Thanks a lot !

ChaoqinRobotics, 2020-07-08:
Hi @narutojxl . Sorry for replying late. I directly follow the imu bias natation convention from *Principles of GNSS, Inertial, and Multisensor Integrated Navigation Systems* written by Paul D Groves. 

Their definitions are:
- acc_w: Accelerometer noise root PSD (micro-g per root Hz, converted to m s^-1.5)                
e.g. IMU_errors.accel_noise_root_PSD = 100 * 9.80665E-6;
- gyr_w: Gyro noise root PSD (deg per root hour, converted to rad s^-0.5)                
e.g. IMU_errors.gyro_noise_root_PSD = 0.01 * deg_to_rad / 60;

narutojxl, 2020-07-08:
It doesn't matter, thanks @qin :)  I will read the reference paper tomorrow.

narutojxl, 2020-07-08:
Hi @qin,
* For acc_w,  convert `ug/sqrt(Hz)` into  `m s^-1.5`,  in code, we do not multiply the front coefficient `100`.
* For gyr_w,  convert `deg/sqrt(hr)` into `rad s^-0.5` in code, we do not multiply the front coefficient `0.01`.

In reference book, it suggest we need to multiply root PSD with sqrt(Hz), do we need to do this in code?
[image]

Liming-Cheng, 2020-12-21:
Hi @narutojxl , 
     Do you solve the problem about imu parameter ? In the UrbanNav dataset, the unit of acc_noise and acc_walk is m/(s^2), and gyro_noise is rad/s [.](https://github.com/weisongwen/UrbanNavDataset/blob/master/UrbanNav-HK-Data20190428/xsens_imu_param.yaml) Should I do any special treatment in the config file?

narutojxl, 2020-12-21:
Hi @Liming-Cheng,
According to author given the reference paper, 
> bias unit  acc_w: Accelerometer noise root PSD (micro-g per root Hz, converted to m s^-1.5), 
   gyr_w: Gyro noise root PSD (deg per root hour, converted to rad s^-0.5)

It seems that above value should be multiply with the root of the sampling rate, when using.
Imu parameters in UrbanNavdataset seems got from `https://github.com/gaowenliang/imu_utils`. I am not sure whether we can directly use these params in lins code, maybe we should do some convertion as you said, look forward author qin can give a explanation :)

fengxingyang, 2021-02-19:
Hi @narutojxl,
I encountered some difficulties in trying to run Lins with the Kitti dataset,the topic of laser_odom_to_init and the topic of aft_mapped_to_init deviate too far in rviz,I have revised the contents of the exp_port.yaml as follows,What's wrong with me?
[image]

[image]