ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #13: doubt about StateEstimator.cpp line 549

xieqi1, 2020-06-13:
Hi,
  I found the code file StateEstimator.cpp line 549
  `updateVec_ = -Kk_ * (residual_ + Hk_ * difVecLinInv_) + difVecLinInv_;`
  a little different from the equation(16) (17) in your paper, I think it seems like this
   `updateVec_ = -Kk_ * (residual_ - Hk_ * difVecLinInv_) + difVecLinInv_;`
 I may have misunderstood, so can you give a explanation. Thank you!

ChaoqinRobotics, 2020-06-13:
Don't forget there is a minus in front of Kk_, while there is NO minus in the paper. If you derive it carefully, you would find that the formula in the paper is exactly the same as our code.

xieqi1, 2020-06-13:
Thank you very much !

closed 2020-06-13 by xieqi1

Gatsby23, 2021-08-16:
> Hi,
> I found the code file StateEstimator.cpp line 549
> `updateVec_ = -Kk_ * (residual_ + Hk_ * difVecLinInv_) + difVecLinInv_;`
> a little different from the equation(16) (17) in your paper, I think it [...]

I'm sorry, I'm still struggling about this code, why this is right? From the equation, I mulplie the minus into the code, the equation from (16) and (17) also should be:
```
updateVec_ = -Kk_ * (residual_ - Hk_ * difVecLinInv_) + difVecLinInv_
```