ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #14: Why not use the covariance from imu pre-integration

xieqi1, 2020-06-13:
It seems that the code don't use the covariance and jacobian calculated by imu pre-integration，and I find the imu only used to get a estimate of pose between two consecutive pointclouds, so why not use the covariance from imu pre-integration as the P in kalman filter? Thank you!

ChaoqinRobotics, 2020-06-13:
Thanks for asking. imu pre-integration is only for state initialization.   We propagate state covariance using IMU in function 'bool predict(double dt, const V3D& acc, const V3D& gyr, bool update_jacobian_ = true)`. The covariance is also updated in this function. ' covariance_ = F_ * covariance_ * F_.transpose() + Gt * noise_ * Gt.transpose();'

Please take a close look at KalmanFilter.hpp. IMU is not only used for pose estimate, but also for covariance update, too, just like a conventional Kalman filter.

xieqi1, 2020-06-13:
Thank you @ChaoqinRobotics, I thought it was in file integrationBase.h, just like VINS-MONO. Thank you for quick answer!

closed 2020-06-13 by xieqi1