ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #16: std::bad_alloc of lins_fusion node

narutojxl, 2020-07-08:
Hi author,
Sometime the  lins_fusion_node will crash with our bag data(over 15G,  not uploaded for now). OS is ubuntu16.04 + gcc7.4.0(g++ 7.4.0) + cmake3.17.2 + PCL1.7.2  The detailed result is :
>
	[ INFO] [1594192703.778903566]: ----> Transform Fusion Started.
	[ INFO] [1594192703.792973732]: ----> LINS Fusion Started.
	[ INFO] [1594192703.807135074]: ----> Feature Extraction Module Started.
	[ INFO] [1594192703.812935854]: Subscribe to ----> /imu/data
	[ INFO] [1594192703.812958958]: Subscribe to ----> /velodyne_points
	[ INFO] [1594192703.826903704]: ----> Map Optimization Started.
	[ INFO] [1594192706.594825669, 1574426489.592443649]: System Converges after 1 iterations
	terminate called after throwing an instance of 'std::bad_alloc'
	  what():  std::bad_alloc
	[lins_fusion_node-6] process has died [pid 21709, exit code -6, cmd /home/jxl/Lins_ws_raw/devel/lib/lins/lins_fusion_node __name:=lins_fusion_node __log:=/home/jxl/.ros/log/1a50bbdc-c0eb-11ea-b3c5-8cec4bc262f3/lins_fusion_node-6.log].
	log file: /home/jxl/.ros/log/1a50bbdc-c0eb-11ea-b3c5-8cec4bc262f3/lins_fusion_node-6*.log

narutojxl, 2020-07-08:
Build lins debug version, in CMakeLists.txt modifiy
`# set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -O3")
set(CMAKE_BUILD_TYPE "Debug")
set(CMAKE_CXX_FLAGS_DEBUG "-std=c++11 -g -O0") ` and then use gdb to launch lins_fusion_node.
When system startup, the map is mess and meaningless.  Termial is always outputing
> [pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.

and the gdb terminal is always output 

> Insufficient matched surfs... 
   Insufficient matched corners...

Any ideas or suggestion ?  How to figure out the crash problem's reason with release version?  Thanks a lot :)

ChaoqinRobotics, 2020-07-08:
Hi @narutojxl . I have not run into this problem before. It seems like that the feature extraction module did not extract enough edge or planar features. Maybe checking the actual number of the edge and planar features at each scan would help.

narutojxl, 2020-07-09:
Hi doctor qin, thanks for your advice.
I add some print information in `LinsFusion::processPointClouds()` func like tuis:
>
	bool LinsFusion::processPointClouds() {
	  
	  std::printf("\n\n");
	  ROS_WARN("prepare for data");

	  // Obtain the next PCL
	  pclBuf_.itMeas_ = pclBuf_.measMap_.upper_bound(estimator->getTime());//第一个时间戳大于滤波器的点云
	  ROS_WARN("PCL---1");
	  sensor_msgs::PointCloud2::ConstPtr pclMsg = pclBuf_.itMeas_->second;
	  scan_time_ = pclBuf_.itMeas_->first;
	  ROS_WARN("PCL---2");
	  distortedPointCloud->clear();
	  ROS_WARN("PCL---3");
	  pcl::fromROSMsg(*pclMsg, *distortedPointCloud);
	  ROS_WARN("obstain PCL");
	  
	  outlierBuf_.itMeas_ = outlierBuf_.measMap_.upper_bound(estimator->getTime());
	  ROS_WARN("outlier---1");
	  sensor_msgs::PointCloud2::ConstPtr outlierMsg = outlierBuf_.itMeas_->second;
	  ROS_WARN("outlier---2");
	  outlierPointCloud->clear();
	  ROS_WARN("outlier---3");
	  pcl::fromROSMsg(*outlierMsg, *outlierPointCloud);
	  ROS_WARN("obstain outlier");

	  cloudInfoBuf_.itMeas_ =
	      cloudInfoBuf_.measMap_.upper_bound(estimator->getTime());
	  ROS_WARN("cloudInfo---1");
	  cloud_msgs::cloud_info cloudInfoMsg = cloudInfoBuf_.itMeas_->second;
	  ROS_WARN("obstain cloudInfo");

	  imuBuf_.getLastTime(last_imu_time_);//得到imu buffer中最新的时间戳
	  if (last_imu_time_ < scan_time_) {
	    ROS_WARN("Wait for more IMU measurement!");
	    return false;
	  }

	  
	  ROS_WARN("process imu data");
	  //process imu code
	  //...
	  //...
	  ROS_WARN("process imu done");
	 
	  //process imu PCL
	  //...
	  //...
	  ROS_WARN("process laser done");
	}

**result sometimes is:**
>
[ WARN] [1594263309.370076439, 1574426578.704153674]: prepare for data
[ WARN] [1594263309.370087625, 1574426578.704153674]: PCL---1
[ WARN] [1594263309.370094189, 1574426578.704153674]: PCL---2
[ WARN] [1594263309.370107002, 1574426578.704153674]: PCL---3
[ WARN] [1594263309.370159534, 1574426578.704153674]: obstain PCL
[ WARN] [1594263309.370168712, 1574426578.704153674]: outlier---1
[ WARN] [1594263309.370177131, 1574426578.704153674]: outlier---2
[ WARN] [1594263309.370188292, 1574426578.704153674]: outlier---3
[ WARN] [1594263309.370214904, 1574426578.704153674]: obstain outlier
[ WARN] [1594263309.370226393, 1574426578.704153674]: cloudInfo---1
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[lins_fusion_node-6] process has died [pid 11305, exit code -6, cmd /home/jxl/Lins_ws/devel/lib/lins/lins_fusion_node __name:=lins_fusion_node __log:=/home/jxl/.ros/log/06205f78-c181-11ea-bd1b-8cec4bc262f3/lins_fusion_node-6.log].
log file: /home/jxl/.ros/log/06205f78-c181-11ea-bd1b-8cec4bc262f3/lins_fusion_node-6*.log
[lidar_mapping_node-7] process has died [pid 11306, exit code -11, cmd /home/jxl/Lins_ws/devel/lib/lins/lidar_mapping_node __name:=lidar_mapping_node __log:=/home/jxl/.ros/log/06205f78-c181-11ea-bd1b-8cec4bc262f3/lidar_mapping_node-7.log].
log file: /home/jxl/.ros/log/06205f78-c181-11ea-bd1b-8cec4bc262f3/lidar_mapping_node-7*.log

**sometimes is:**
>
[ WARN] [1594263484.332811216, 1574426582.324104907]: prepare for data
[ WARN] [1594263484.332858204, 1574426582.324104907]: PCL---1
[ WARN] [1594263484.332870284, 1574426582.324104907]: PCL---2
[ WARN] [1594263484.332883689, 1574426582.324104907]: PCL---3
[ WARN] [1594263484.333023025, 1574426582.324104907]: obstain PCL
[ WARN] [1594263484.333035412, 1574426582.324104907]: outlier---1
[lins_fusion_node-6] process has died [pid 14345, exit code -11, cmd /home/jxl/Lins_ws/devel/lib/lins/lins_fusion_node __name:=lins_fusion_node __log:=/home/jxl/.ros/log/06205f78-c181-11ea-bd1b-8cec4bc262f3/lins_fusion_node-6.log].
log file: /home/jxl/.ros/log/06205f78-c181-11ea-bd1b-8cec4bc262f3/lins_fusion_node-6*.log

I guess maybe we need  `std::mutex` when using  the three variable `pclBuf_`,  `outlierBuf_` and `cloudInfoBuf_`. I will try later.

narutojxl, 2020-07-09:
Figure out the problem's reason. Sometimes the return value is`end()`.
> pclBuf_.itMeas_ = pclBuf_.measMap_.upper_bound(estimator->getTime());
outlierBuf_.itMeas_ = outlierBuf_.measMap_.upper_bound(estimator->getTime());
cloudInfoBuf_.itMeas_ = cloudInfoBuf_.measMap_.upper_bound(estimator->getTime());

When return value is `end()`, maybe we should, the same for outlierBuf_ and cloudInfoBuf_.
>
	if(pclBuf_.itMeas_ == pclBuf_.measMap_.end()){
	    //ROS_WARN("Not exist reasonable pcl\n");
	    pclBuf_.clean(estimator->getTime());
	    return false;
	  }

closed 2020-07-09 by narutojxl