ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #18: How to get IMU noise parameter?

LeisureLei, 2020-07-27:
I want to run LINS in my own platform. But I don't konw how to get IMU noise parameter. And I am still confused with the parameter units.

ChaoqinRobotics, 2020-07-29:
Thanks for your question. You can take a look at this file `lins/include/parameters.h` where the definitions of all IMU noise parameters are listed. And `lins/include/KalmanFilter.hpp` shows how these parameters are used. 

For example, some definitions are:
acc_w: Accelerometer noise root PSD (micro-g per root Hz, converted to m s^-1.5)
e.g. IMU_errors.accel_noise_root_PSD = 100 * 9.80665E-6;
gyr_w: Gyro noise root PSD (deg per root hour, converted to rad s^-0.5)
e.g. IMU_errors.gyro_noise_root_PSD = 0.01 * deg_to_rad / 60;

LeisureLei, 2020-07-30:
> Thanks for your question. You can take a look at this file `lins/include/parameters.h` where the definitions of all IMU noise parameters are listed. And `lins/include/KalmanFilter.hpp` shows how [...]

Thanks four your reply. I have another question now. According to README, "imu_misalign_angle" is the yaw angle between LIDAR and IMU. And I find that "imu_misalign_angle" is the yaw angle between IMU frame and Vehicle frame in code. Is Vehicle frame equal to LIDAR frame? Also, at the end of README, you mean that the extrinsic parameters between LIDAR and IMU can be set zero but "imu_misalign_angle" in config file is equal to 3. That's what confused me.