ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #19: Question about the performance of the code and performance in the vedio.

Gaochao-hit, 2020-07-29:
Thank you for your great work. I have a quetion whether the code you release is the same as that in the vedio. The performance of your algorithm in the vedio is really excellent.

ChaoqinRobotics, 2020-07-29:
It is a simplified version of the original code, but the performance is bascially the same. Please feel free to test it on your dataset. One thing you need to notice is that since the IMU-lidar extrinsic parameters are not considered in this version of LINS, it is suggested that the IMU should be placed as close as possible to the Lidar.

Gaochao-hit, 2020-07-29:
> It is a simplified version of the original code, but the performance is bascially the same. Please feel free to test it on your dataset. One thing you need to notice is that since the IMU-lidar [...]

Thank you for your reply.

closed 2020-07-29 by Gaochao-hit