ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #21: Doubt on equation (18) in the paper

Tompson11, 2020-08-13:
Hi @ChaoqinRobotics :
Thank you for sharing your work!
I have doubt on equation (18) in your paper:
[image]
how did you get that? According to the equation (54) in the reference [[29]](https://www.research-collection.ethz.ch/bitstream/handle/20.500.11850/263423/ROVIO.pdf?sequence=1&isAllowed=y), I think it should be:
[image]

ChaoqinRobotics, 2020-08-13:
This is call the symmetric and positive Joseph form P= (I-KH)P(I-KH)+KVK. Because the original one P=(1-KH)P is known to have poor numerical stability, as its outcome is not guaranteed to be symmetric nor positive denite.

Tompson11, 2020-08-13:
Thanks a lot! I've never seen this form before and I'll learn more about it ~