ChaoqinRobotics/LINS---LiDAR-inertial-SLAM #27: IMU parameters

Liming-Cheng, 2020-12-21:
Dear @ChaoqinRobotics ,
      When I test LINS on UrbanNav dataset, I'm confused the imu parameter in LINS's config file.  In the UrbanNav dataset, the unit of acc_noise and acc_walk is m/(s^2), and gyro_noise is rad/s. Should I do any special treatment in LINS's config file?
Here is imu parameters in UrbanNavdataset https://github.com/weisongwen/UrbanNavDataset/blob/master/UrbanNav-HK-Data20190428/xsens_imu_param.yaml.  Can you help me with that ? Thanks a lot!

ChaoqinRobotics, 2020-12-23:
Thanks for your question. A brief answer is I am not sure about it. And a long answer is that I suggest you to apply the strategy of trials and tunnings of these noise parameters. Here is the reason. For optimal estimators such as Kalman filters and batch estimators, the exact values of the noise DOES NOT matter. What really matter is the RATIO between the process noise (e.g. Q matrix) and the measurement noise (e.g. R matrix). Sometimes the correct IMU noise parameters do not yield the best performance because of the uncertainty of the measurement noise. Looking for a trade-off might be a more practical solution.

Liming-Cheng, 2020-12-26:
Thanks for your reply!  But what I really want to know is  whether we need to change the unit of Urbannav's imu parameters. And how to translate the unit to be consistent with LINS's imu parameters.

ChaoqinRobotics, 2020-12-26:
Sorry, I don't know, either.

bryantaoli, 2021-06-27:
> Dear @ChaoqinRobotics ,
> When I test LINS on UrbanNav dataset, I'm confused the imu parameter in LINS's config file. In the UrbanNav dataset, the unit of acc_noise and acc_walk is m/(s^2), and [...]

I also tested LINS on UrbanNav and UrbanLoco and had some problems, some of which are the same as yours. Maybe we can talk about it